FujiNetWIFI/fujinet-firmware #597: Changing disk mounts from webui prevents disk rotation on Atari

mozzwald, 2023-04-29:
Steps to reproduce:

1. Mount DOS disk to slot 1, all other slots empty
2. Boot to DOS
3. Mount disk to slot 2 and 3 from webui
4. press disk rotate button (button A) and disks are not rotated

The file handles are reset when mounting from webui and not recreated in `populate_slots_from_config()` and the disk rotate function checks for file handles being null.

This attempt at recreating the file handle sometimes worked, but not always and I'm not sure why. Maybe there's another way to do it. 

```
void sioFuji::_populate_slots_from_config()
{
    char flag[3] = {'r', 0, 0};

    for (int i = 0; i < MAX_HOSTS; i++)
    {
        if (Config.get_host_type(i) == fnConfig::host_types::HOSTTYPE_INVALID)
            _fnHosts[i].set_hostname("");
        else
            _fnHosts[i].set_hostname(Config.get_host_name(i).c_str());
    }

    for (int i = 0; i < MAX_DISK_DEVICES; i++)
    {
        _fnDisks[i].reset();

        if (Config.get_mount_host_slot(i) != HOST_SLOT_INVALID)
        {
            if (Config.get_mount_host_slot(i) >= 0 && Config.get_mount_host_slot(i) <= MAX_HOSTS)
            {
                strlcpy(_fnDisks[i].filename,
                        Config.get_mount_path(i).c_str(), sizeof(fujiDisk::filename));
                _fnDisks[i].host_slot = Config.get_mount_host_slot(i);
                if (Config.get_mount_mode(i) == fnConfig::mount_modes::MOUNTMODE_WRITE)
                {
                    _fnDisks[i].access_mode = DISK_ACCESS_MODE_WRITE;
                    flag[1] = '+';
                }
                else
                    _fnDisks[i].access_mode = DISK_ACCESS_MODE_READ;

                // Open the file
               _fnDisks[i].fileh = _fnHosts[Config.get_mount_host_slot(i)].file_open(_fnDisks[i].filename, _fnDisks[i].filename, sizeof(_fnDisks[i].filename), flag);
            }
            else
                Debug_printf("populate_slots_from_config: slot %d out of range\n", i);
        }
        else
            Debug_printf("populate_slots_from_config: invalid host slot %d\n", i);
    }
}
```

dillera, 2026-10-03:
This issue has had no activity for about 3 years, and there have been 2430 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-03:
Still happens on current master. The old reopen attempt is gone, so `populate_slots_from_config()` now resets every slot's `fileh` and reopens nothing. After one web UI mount even the slots mounted from CONFIG have a null `fileh`, so `fujicmd_image_rotate()` rotates nothing. The disks keep working because the media objects stay mounted. Rotate could check whether the slot's disk device is mounted instead of `fileh`, or the web UI path could skip slots that are already mounted.

dillera, 2026-10-03:
Thanks for confirming this still happens on current firmware — keeping it open.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

closed 2026-10-14 by tschak909