FujiNetWIFI/fujinet-firmware #600: SIO highspeed problem (Atari)

XeAtari, 2023-05-06:
Hi,

Since some time ago I had the feeling that there was some kind of problem when using highspeed SIO routines.

As you will know a highspeed SIO routine queries a speed byte from the drive in normal speed and then sends succeeding read or write commands in highspeed.

Test environment:

Atari 130 XE (PAL) with FujiNet connected, nothing else on the SIO, stock OS
SD-Card in FujiNet
Drives 1-8 mounted with ATRs on SD card
Drive 1 contains SpartaDOS 33b (uses US Doubler SIO for highspeed)
SIO-speed: 9

Steps:

Boot DOS -> OK
DIR D1: -> OK
DIR D2: -> OK
DIR D3: -> OK
DIR D4: -> OK
DIR D5: -> Timeout
DIR D6: -> OK
DIR D5: -> OK (D5 now works)
DIR D7: -> Timeout
DIR D8: -> OK
DIR D7: -> OK (D7 now works)

If I use a different order to call drives then different drives time out.

But if I go 1-2-3-4-5 it is always D5 that times out.

If I go 1-2-3-4-6, no 6 does not time out. Weird.

If I perform the procedure a second time without resetting the Atari drives do not time out. This will be because the Atari already knows the highspeed value for each drive and does not query it again. So there is no mixture of normal and highspeed on the SIO bus.

If I perform the very same test procedure running APE (registered version) or RespeQT, the problem does not occur.

Attached please find flasher logs.

See issue #549, maybe that is related.

The problem is not related to the beta firmware used, I guess.

[HighSpeed-Problem.TXT](https://github.com/FujiNetWIFI/fujinet-platformio/files/11411698/HighSpeed-Problem.TXT)

XeAtari

tschak909, 2023-07-25:
Well, nevermind, I was able to reproduce it with the dupdsk problem in SpartaDOS X. This is weird.

tschak909, 2023-07-25:
I have a potential fix, can you try this build zip with the fujinet flasher?

XeAtari, 2023-07-28:
> completed via d377d69

Hi,

thanks a lot for looking into this.

Unfortunately I cannot compile firmwares myself so I'll have to wait for 
a release.

In case you still need SpartaDOS 3.3b please tell me. I'll send an ATR.

==

XeAtari, 2023-08-21:
Hello Thom,

I installed v1.1.0 (2023-08-13) only a few minutes ago.

I extensively ran DIR commands using SD33B across all 8 drives.

I did no longer encounter the problem so it looks like you managed to 
solve things. Great job!

Best, XeAtari

==

commit d377d695 (2023-07-25): [sio][disk] fixes #600
Preemptively switch baud rate on sio get high speed.
closed 2024-08-24 by tschak909