FujiNetWIFI/fujinet-firmware #613: [atari][network] Translation not honored on transmit.

tschak909, 2023-06-09:
When transmitting data via the network device, the Atascii EOL is not translated to the desired line ending.

e.g. given the following BASIC:
```
10 OPEN #1,12,2,"N:TCP://1.2.3.4:6502/":REM DO LF TRANSLATION
20 PRINT #1;"HELLO"
30 CLOSE #1
```

What should be received on the other end, should be:

```
HELLO<lf aka 0x0a>
```

However what is received is actually:

```
HELLO<eol aka 0x9b>
```

This is a potential problem in the Protocol base class:
https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/network-protocol/Protocol.cpp

markjfisher, 2023-06-11:
I can't reproduce this. I'm using a local build of fujinet from latest master on my device (commit: 5cc6bdb9).

tcpdump output:
```
09:40:17.957237 IP Fujifish.home.53232 > unifi.8000: tcp 16
	0x0000:  4500 0038 0013 0000 ff06 3785 c0a8 0182  E..8......7.....
	0x0010:  c0a8 0155 cff0 1f40 9514 8138 4333 7953  ...U...@...8C3yS
	0x0020:  5018 fffe e554 0000 2020 2020 2020 2020  P....T..........
	0x0030:  2020 4845 4c4c 4f0a
```
which shows the mode translation is working.

I tried all modes (0 = none [0x9b], 1 = CR [0x0d], 2 = LF [0x0a], 3 = CRLF [0x0d0a]) and all worked as expected.

markjfisher, 2023-06-11:
Tested with following setup (in case I come back to this in a year and wonder how I did it):

## Linux laptop on network

### Terminal 1 - tcpdump listener on port 8000 of wifi device:
```
sudo tcpdump -Xq -i wlp59s0 port 8000
```

### Terminal 2 - netcat listener (note, ncat for -k flag)
```
ncat -l -k 192.168.1.85 8000
```

### Atari:
1. Boot with nhandler atr file.
2. Run program as outlined in the OP

closed 2023-07-25 by tschak909