FujiNetWIFI/fujinet-firmware #644: Add proper handling for HAPPY commands.

tschak909, 2023-07-26:
Currently, with U1MB's HSIO support enabled, certain conditions can cause the HSIO code to emit a HAPPY READ ($72) command.

A decision needs to be made whether to support HAPPY commands, or to try and prevent the computer from thinking they can be emitted (this causes issues e.g. with ATX)

dillera, 2026-10-03:
This issue has had no activity for about 3 years, and there have been 2000 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-16:
@tschak909 FujiNet NAKs every Happy opcode (`$48`, `$72`, …) at the command frame, so U1MB's HSIO can't be detecting a Happy by ACK; it must be keying on something else, probably the `$3F` index we return (`hsioindex`) or a drive heuristic in that HSIO build. Do you remember which index you were running, and does it still happen on current firmware? If it's the index, the fix is a documented value to avoid, not Happy emulation. If we can't pin it down I'd close this as not reproducible rather than add Happy commands.