FujiNetWIFI/fujinet-firmware #690: Add BASIC and DIGEST auth support to HTTP protocol adapter.

tschak909, 2023-12-18:
Given the following BASIC code
```
XIO #8,253,0,0,U$
XIO #8,254,0,0,P$
NOPEN 8,4,0,URL$
... it feeds me "no username or password provided." 

U$="N:my_username"
P$="N:mypassword"
URL$=working url clearly because I am getting the response
```

or via the URL

```
N:http://username:password@host.domain/path/?...
```

Be able to fill out the auth_type field in esp_http_client() so that the appropriate BASIC or DIGEST request can be made for HTTP endpoints that require authentication in those two methods.

Relevant changes would need to be made in the fnHTTPClient:
https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/http/fnHttpClient.cpp#L45

and in the HTTP protocol adapter:
https://github.com/FujiNetWIFI/fujinet-platformio/blob/master/lib/network-protocol/HTTP.cpp

with one additional command added to set the AUTH type.

dillera, 2026-10-03:
This issue has had no activity for about 3 years, and there have been 1547 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-11:
Still valid for the XIO 253/254 path: `lib/network-protocol/HTTP.cpp` never reads the protocol's login/password, so neither HTTP client gets them. The URL form (`N:http://user:pass@host/...`) does work now: on ESP32 the vendored `fn_esp_http_client` takes the credentials and answers a 401 with BASIC or DIGEST, and on FujiNet-PC `mgHttpClient` sends BASIC (no DIGEST there). The SSH adapter already merges URL and XIO credentials, so HTTP could do the same.