FujiNetWIFI/fujinet-firmware #716: Add ability to do raw data POST

tschak909, 2024-04-11:
from discord thread:

```
curl --location 'http://192.168.2.69’ \
--header 'Content-Type: application/json' \
--data-raw '{
    “Namne: “Paulo “Garcia,
    “Computer”: “Atari 800
}'
```

pedgarcia, 2024-06-24:
Hey @tschak909 , anyway I can help with this? I haven't look too deeply into the code, but maybe if you point me out in the right direction, I might be able to help with this issue.

Cheers

tschak909, 2024-06-24:
This should give you a trail to show how commands wind up at the network device, and the protocol adapter:

https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/src/main.cpp#L502
https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/lib/bus/sio/sio.cpp#L402
https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/lib/device/sio/network.cpp#L859
https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/lib/device/sio/network.cpp#L645
https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/lib/network-protocol/HTTP.cpp#L78

Hopefully this helps.

We can answer questions as you have them, in the discord: https://discord.gg/2Ce9guX :)

-Thom

pedgarcia, 2024-06-24:
Perfect, thanks. Let's see if I can un-rust my C :)

dillera, 2026-10-03:
This issue was opened about 2 years ago, and the code it refers to has changed since.

- `lib/device/sio/network.cpp` no longer exists in the codebase

The files it names have changed in 103 commit(s) since then.

Closing as no longer applicable to the current firmware. If you can still reproduce this on a current firmware build, please reopen it (or open a new issue) with the firmware version, the platform, and the steps.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

closed 2026-10-03 by dillera