FujiNetWIFI/fujinet-firmware #732: "Enable CONFIG boot disk" not working on Apple

cathrynm, 2024-05-13:
I don't think Enable CONFIG Boot disk turned off is working on Fujinet for Apple.  When I disable this, reset and reboot, I still see a drive called 'Config' mounted in GS.OS.  If I 'Mount All' it goes back to my selected hard drives.  Basically from GS.OS, config isn't hugely useful anyway since it's mostly built around rebooting the PC. it'd be nice if it just always starts up with whatever drives were  mounted from the web UI, which I think is what this option is supposed to do.  (I'm not mear my Atari, so I can't compare right now.)

dillera, 2026-10-03:
This issue has had no activity for about 2 years, and there have been 1151 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-11:
@cathrynm Fixed by #1193 (Feb 2026): the Apple setup now honors "Enable CONFIG boot disk" and mounts the disks you set up in the web UI when it's off. Yappyfox confirmed the same in #1125. Closing.

closed 2026-10-11 by RichStephens