FujiNetWIFI/fujinet-firmware #808: Can't Build Flashsystem System using VS Code and Windows

venice1200, 2024-10-15:
At frist, many thanks for this great piece of hard- and software.

I don't know if it's a setup issue at my end or something related to the firmware code.
Maybe someone can give me some help.

To build the FujiNet binaries for the **"fujinet-devkit-shield"** from scratch I added a new user to my Windows 10 system 
and setup VS Code as shown in the FujiNet wiki.

I can compile the code but if I want to build the filesystem image I get the follwing warning/error under Windows...

```
|-- config
|-- http
Building in debug mode
Building FS image from 'E:\E:\Projekte\PlattformIO\Projects\fujinet-firmware\data\BUILD_ATARI' directory to .pio\build\fujinet-v1-8mb\littlefs.bin
warning: can't read source directory
*** [.pio\build\fujinet-v1-8mb\littlefs.bin] Error 1
============================================================================================= [FAILED] Took 30.59 seconds =============================================================================================
Environment     Status    Duration
--------------  --------  ------------
fujinet-v1-8mb  FAILED    00:00:30.592
======================================================================================== 1 failed, 0 succeeded in 00:00:30.592 ========================================================================================
```

The Drive Letter, here **"E:\"**, is doubled in the Path shown in the Line "**Build FS image from**".
Is this something I can set/adjust?

If I use Linux it works!

mozzwald, 2024-10-15:
@venice1200 the issue is with this variable in `platformio.ini`:

`data_dir = /$PROJECT_DIR/data/${fujinet.build_platform}`

You can edit it to the exact path, and/or remove the first slash, and/or replace with the correct slashes for windows

venice1200, 2024-10-19:
Without the first slash it works.

Many Thanks!