FujiNetWIFI/fujinet-firmware #818: Cant get WebDav to work

sideburn, 2024-12-09:
I have updated to 1.4 and when i go to /dav through all of the methods none of them load a page it is just blank or cannot connect. I am able to get to the web ui no problem. just adding /dav doesn't work. Tried safari and chrome on mac and Edge on windows. AM I missing something?

mozzwald, 2024-12-15:
@sideburn try using a proper webdav client like https://cyberduck.io or similar

mozzwald, 2024-12-25:
> Thank you! This is awesome. Been looking for a way to manage the SD card without having to remove it and this does everything I need. Gave the device a hostname and i'm good to go even if the ip changes, sweet! :)

Great! Glad you got it going

closed 2024-12-25 by mozzwald