FujiNetWIFI/fujinet-firmware #827: [network-protocol][tnfs] Reading after Opening existing file for Update yields 0 bytes waiting.

tschak909, 2025-01-24:
Opening an existing with 12 (update) and reading from the file yields 0 bytes waiting.

dillera, 2026-10-03:
This issue has had no activity for about 21 months, and there have been 797 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-11:
@tschak909 Fixed by fc17f0346 ("do not emit EOF on write", the morning after this was filed): `open_file()` now calls `stat()` for update mode, so `fileSize` is the file's real size instead of 0. Still that way on master. Closing.

closed 2026-10-11 by RichStephens