FujiNetWIFI/fujinet-firmware #858: Put out v1.5 release

FozzTexx, 2025-03-12:
Put out the v1.5 release. An RC branch has been created: https://github.com/FujiNetWIFI/fujinet-firmware/tree/v1.5-rc

Platforms that should be working for v1.5:
* Atari 8
* Adam
* Apple II
* CoCo

Apple II testing has revealed some glitches with Disk II, being tracked here: https://github.com/FujiNetWIFI/fujinet-firmware/issues/855

nwah, 2025-03-15:
Minor display bug in Apple2 Config:
1. Start with empty Disk II drives
2. Highlight Disk II slot and press `E` to eject
3. Only first letter of drive label remains (`S` or `N` instead of `S6D2` or `NA`)
4. Tab back to host list and drive labels go back to normal

[image]
[image]
[image]

FozzTexx, 2025-04-02:
v1.5.0 tag has been created!

closed 2025-04-02 by FozzTexx