FujiNetWIFI/fujinet-firmware #866: Disk II write has insufficient DMA RAM to start SPI continuous capture

FozzTexx, 2025-03-19:
More often than not it seems like Disk II write doesn't work because when the SPI continuous needs to start there is not enough DMA RAM remaining to allocate the circular buffer. About 60k of DMA RAM is needed.

`19:25:19.933 > SPI_SMA: Insuffucient RAM 51495 < 53248 + 4992`

Pre-allocating enough RAM during startup means there's not enough RAM to use the SD card.

Investigating where DMA RAM is going. Current findings:

* There's almost 200k of DMA RAM available at startup
* WiFi eats about 80k of DMA RAM.
* Disk II pre-loaded image uses PSRAM, not DMA RAM
* When write mode is activated there is only about 50k available.

tschak909, 2025-03-19:
I guess it's (find a bit here, find a bit there...) time...
-Thom

FozzTexx, 2025-03-19:
> I guess it's (find a bit here, find a bit there...) time...

What's the best way to do that? I've been trying to figure out where the RAM is going but it's not easy.

FozzTexx, 2025-03-23:
Fixed in https://github.com/FujiNetWIFI/fujinet-firmware/pull/873

closed 2025-03-23 by FozzTexx