FujiNetWIFI/fujinet-firmware #867: Booting ProDOS 2.4.2 from Disk II when write is enabled causes Relocation error

FozzTexx, 2025-03-19:
Reported by ppuskari on Discord.

When ProDOS is booted from CONFIG it starts to boot then fails with Relocation error.

Steps:

* Reset FujiNet
* Boot into CONFIG and put ProDOS 2.4.2 disk into Disk II D1 slot
* ESC from CONFIG and let computer reboot
* ProDOS will fail with Relocation error

ppuskari reports this happens on a IIe with a Dsub19 IO Controller card.

robjustice, 2025-03-20:
I tested this with a enhanced //e with a ramworks3 card, db19 disk controller in s6, grappler minus softsp card in slot 5.
Built from branch v1.5-rc + added the latest config into this.

Mounted prodos 2.4.2.po as writable in the S6D1 slot. Nothing in SP slots

Rebooted, and got the 'Relocation/Configuration Error' Seems to do it every time. No difference if usb is connected or not.

Logs and disk used attached.

[ProDOS_2_4_2.zip](https://github.com/user-attachments/files/19361198/ProDOS_2_4_2.zip)

[reloc_error.log](https://github.com/user-attachments/files/19361211/reloc_error.log)

FozzTexx, 2025-03-20:
> I tested this with a enhanced //e with a ramworks3 card, db19 disk controller in s6, 

Are you able to test with a Disk II controller instead of the Dsub19 IO controller?

FozzTexx, 2025-03-20:
> Built from branch v1.5-rc

What's the exact commit ID the FN shows, either at reset or on the web page?

`07:39:15.594 > FujiNet v1.5-rc1-2-g27afdbb6 2025-03-18 06:59:39 (APPLE) Started @ 7
`

[image]

robjustice, 2025-03-20:
Can you guide me how to get that specific version pulled down?

FozzTexx, 2025-03-20:
> Can you guide me how to get that specific version pulled down?

`git checkout 98924f8`

robjustice, 2025-03-20:
i checked that one out above and still the same, see attached log. I must be missing something to put the commit id in the version.

[reloc_error_2.log](https://github.com/user-attachments/files/19376245/reloc_error_2.log)

And i'm a bit confused on the commit id, the one at the top is different to this one.

FozzTexx, 2025-03-20:
> i checked that one out above and still the same, see attached log. I must be missing something to put the commit id in the version.

You need to turn on the debug_version.py in platformio.local, see the main ticket for v1.5. https://github.com/FujiNetWIFI/fujinet-firmware/issues/858

> And i'm a bit confused on the commit id, the one at the top is different to this one.

The version in the screenshot was just an example of how the version should be shown on the FN

robjustice, 2025-03-28:
I tested this again with this version:  v1.5-rc1-26-g136a63a3 
I don't get the relocation error, and did a bit of playing with 2 disks loaded, etc. and have not seen it happen during my testing today

tschak909, 2025-03-30:
I can boot and use ProDOS, mounted r/w on disk ][ slot s6d1 with commit 98924f8 .

FozzTexx, 2025-03-30:
Nobody is able to reproduce anymore.

closed 2025-03-30 by FozzTexx