FujiNetWIFI/fujinet-firmware #868: Mounting Disk II image causes heap corruption

FozzTexx, 2025-03-19:
While trying to track down where RAM is being used in https://github.com/FujiNetWIFI/fujinet-firmware/issues/866 I turned on `heap_caps_check_integrity_all(1)` in the loop in main.cpp. With a disk configured in Disk II Drive 1, clicking `MOUNT ALL DISKS` in the web interface will mount the disk and then immediately cause a Guru error on the FujiNet firmware.

FozzTexx, 2025-03-20:
It seems this line in mediaTypeDSK.cpp is causing heap corruption:

`        dest[6653] = 10;`

If I `#ifdef` it out then the heap corruption right after mounting the disk stops. Not sure if it's writing past the end of the buffer or something else is using that value to do something that mangles a buffer.

tschak909, 2025-03-20:
Why the hell are we writing past a 6646 byte buffer?
-Thom

FozzTexx, 2025-03-20:
> Why the hell are we writing past a 6646 byte buffer?

Yet another case of hardcoded numeric values with no constant defined so it's not clear what the data represents or why it's needed.