FujiNetWIFI/fujinet-firmware #874: [apple2] WDT timeout when writing from one Disk ][ image to another.

tschak909, 2025-03-22:
Setup:
Apple ][ plus with SoftSP v6 in Slot 5
Apple ][ DB19 controller in Slot 6.

Procedure:

(1) mount DOS 3.3 MASTER in S6,D1
(2) mount blank DOS 3.3 disk prepared with CiderPress in S6,D2 (I stored on my local TNFS)
(3) BOOT
(4) BRUN FID
(5) COPY FILES
(6) COPY SOURCE S6,D1
(7) COPY DEST S6,D2
(8) FILENAME =
(9) PROMPTING? NO.
(10) INSERT DISKS
(11) Wait for copy to complete

For me it failed during copy of BIORHYTHM, with the following crash:

```
19:08:41.668 > Copy track: 73
19:08:41.680 > Copy track: 69
19:08:41.691 > Copy track: 65
19:08:41.702 > Copy track: 61
19:08:41.714 > Copy track: 57
19:08:41.725 > Copy track: 53
19:08:41.736 > Copy track: 49
19:08:41.748 > Copy track: 45
19:08:41.759 > Copy track: 41
19:08:41.770 > Copy track: 37
19:08:41.782 > Copy track: 33
19:08:41.793 > Copy track: 29
19:08:41.804 > Copy track: 25
19:08:41.816 > Copy track: 21
19:08:41.830 > Copy track: 17
19:08:41.847 > Copy track: 13
19:08:41.862 > Copy track: 12
19:08:41.862 > trk pos 012 on d2
19:08:41.862 > Disk II iwm queue receive 2944 31938 34754 50304
19:08:41.863 > Disk II write Qtrack/sector: 96/10  bit_len: 2816
19:08:41.868 > Disk II used: 2944
19:08:41.868 > Disk II sector data: 10
19:08:41.868 > DSK writing track 24 sector 10 
19:08:41.997 > Copy track: 15
19:08:42.014 > Copy track: 19
19:08:42.027 > Copy track: 23
19:08:42.039 > Copy track: 27
19:08:42.051 > Copy track: 31
19:08:42.062 > Copy track: 35
19:08:42.074 > Copy track: 39
19:08:42.077 > Copy track: 40
19:08:42.077 > trk pos 040 on d2
19:08:42.077 > Disk II iwm queue receive 2944 38338 41154 50304
19:08:42.078 > Disk II write Qtrack/sector: 96/12  bit_len: 2816
19:08:42.084 > Disk II used: 2944Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 
19:08:42.894 > 
19:08:42.894 > Core  1 register dump:
19:08:42.894 > PC      : 0x40098448  PS      : 0x00060d35  A0      : 0x80096b73  A1      : 0x3ffc2d30  
19:08:42.895 > A2      : 0x3ffbd058  A3      : 0x3ffbcbdc  A4      : 0x00000004  A5      : 0x00060d23  
19:08:42.898 > A6      : 0x000000fd  A7      : 0x0000abab  A8      : 0x3ffbcbdc  A9      : 0x00000008  
19:08:42.899 > A10     : 0x3ffbcbdc  A11     : 0x00000008  A12     : 0xb33fffff  A13     : 0x00060d23  
19:08:42.902 > A14     : 0x3ffd9c5c  A15     : 0x0000abab  SAR     : 0x00000013  EXCCAUSE: 0x00000006  
19:08:42.903 > EXCVADDR: 0x00000000  LBEG    : 0x4008f7c0  LEND    : 0x4008f7dc  LCOUNT  : 0x00000000  
19:08:42.906 > Core  1 was running in ISR context:
19:08:42.906 > EPC1    : 0x4012ddbf  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000
19:08:42.908 > 
19:08:42.909 > 
19:08:42.909 > Backtrace: 0x40098445:0x3ffc2d30 0x40096b70:0x3ffc2d50 0x40095aae:0x3ffc2d70 0x40123226:0x3ffc2db0 0x40123ca2:0x3ffc2df0 0x401058bf:0x3ffc2e20 0x40082b71:0x3ffc2e70 0x40082a41:0x3ffc2e90 0x400834d9:0x3ffc2eb0 0x400861c1:0x3ffc2ef0 0x4000bfed:0x3ffd49d0 0x4009880d:0x3ffd49e0 0x40095a14:0x3ffd4a00 0x401232cb:0x3ffd4a40 0x40123ca2:0x3ffd4a80 0x401058bf:0x3ffd4ab0 0x40081aaf:0x3ffd4b00 0x40081fa8:0x3ffd4cb0 0x400d2ef0:0x3ffd4cd0 0x400984f5:0x3ffd4cf0
19:08:43.300 > 
[... 232 lines cut ...]
19:08:44.858 > trk pos 081 on d2
19:08:44.860 > trk pos 082 on d2
19:08:44.863 > trk pos 083 on d2
19:08:44.866 > trk pos 084 on d2
19:08:44.869 > trk pos 085 on d2
19:08:44.872 > trk pos 086 on d2
19:08:44.875 > trk pos 087 on d2
19:08:44.878 > trk pos 088 on d2
19:08:44.882 > trk pos 089 on d2
19:08:44.885 > trk pos 090 on d2
19:08:44.889 > trk pos 091 on d2
19:08:44.893 > trk pos 092 on d2
19:08:44.897 > trk pos 093 on d2
19:08:44.901 > trk pos 094 on d2
19:08:44.906 > trk pos 095 on d2
19:08:44.906 > trk pos 096 on d2WIFI_EVENT_STA_CONNECTED
19:08:46.825 > 
19:08:46.825 > disk ii enable states: 00
19:08:46.825 > stop diskII
19:08:46.825 > disk ii enable states: 00IP_EVENT_STA_GOT_IP
19:08:46.987 > Obtained IP address: 192.168.1.16
19:08:46.987 > SNTP client start
19:08:46.989 > Starting web server on port 80
19:08:50.638 > SNTP time sync event: Sat Mar 22, 00:08:50 2025 +0000
```

tschak909, 2025-03-22:
No longer crashes if I apply the following patch:
```patch
diff --git a/lib/device/iwm/disk2.cpp b/lib/device/iwm/disk2.cpp
index 4207370d..d5623c10 100644
--- a/lib/device/iwm/disk2.cpp
+++ b/lib/device/iwm/disk2.cpp
@@ -149,7 +149,7 @@ void IRAM_ATTR iwmDisk2::change_track(int indicator)
         ((MediaTypeWOZ *)_disk)->track_len(track_pos),
         ((MediaTypeWOZ *)_disk)->num_bits(track_pos),
         NS_PER_BIT_TIME * ((MediaTypeWOZ *)_disk)->optimal_bit_timing);
-    Debug_printf("\nCopy track: %d", track_pos);
+    //Debug_printf("\nCopy track: %d", track_pos);
   }
   else
     diskii_xface.copy_track(
```

FozzTexx, 2025-03-22:
Does disabling the one that caused it to crash not fix it?

`19:08:43.300 >   #16 0x40081aaf:0x3ffd4b00 in iwmBus::serviceDiskIIWrite() at lib/bus/iwm/iwm.cpp:671`

tschak909, 2025-03-22:
That also works.

FozzTexx, 2025-03-23:
Fixed by https://github.com/FujiNetWIFI/fujinet-firmware/pull/877

closed 2025-03-23 by FozzTexx