FujiNetWIFI/fujinet-firmware #884: [v1.5-rc] DOS 3.3 FID reports writable Disk ][ image being write protected

tschak909, 2025-03-23:
Test procedure:

* Mount DOS 3.3 in S6D1 (R)
* Create new disk for S6D2 (W)
* Boot
* SAVE FOO,S6,D2 (works)
* DELETE FOO,S6,D2 (works)
* BRUN FID,S6,D1
* COPY FILES
* Source S6D1
* Dest S6D2
* Filename =
* PROMPTING? (N)
* COPYING FILE HELLO
DISK WRITE PROTECTED

wtf?

-Thom

tschak909, 2025-03-23:
A similar issue presents itself trying to use COPY II PLUS to copy either FILES or DISK (without formatting), D2 shows up as WRITE PROTECTED.

tschak909, 2025-03-23:
Showing when it happens from the firmware side. 

```
13:04:34.345 > disk ii enable states: 00
13:07:32.476 > disk ii enable states: 01
13:07:32.476 > disk ii enable states: 01
13:07:32.476 > start diskII d1
13:07:32.628 > trk pos 018 on d1
13:07:32.640 > trk pos 019 on d1
13:07:32.644 > trk pos 020 on d1
13:07:32.649 > trk pos 021 on d1
13:07:32.653 > trk pos 022 on d1
13:07:32.657 > trk pos 023 on d1
13:07:32.660 > trk pos 024 on d1
13:07:32.664 > trk pos 025 on d1
13:07:32.667 > trk pos 026 on d1
13:07:32.671 > trk pos 027 on d1
13:07:32.673 > trk pos 028 on d1
13:07:32.676 > trk pos 029 on d1
13:07:32.679 > trk pos 030 on d1
13:07:32.682 > trk pos 031 on d1
13:07:32.685 > trk pos 032 on d1
13:07:32.688 > trk pos 033 on d1
13:07:32.690 > trk pos 034 on d1
13:07:32.694 > trk pos 035 on d1
13:07:32.696 > trk pos 036 on d1
13:07:32.699 > trk pos 037 on d1
13:07:32.702 > trk pos 038 on d1
13:07:32.705 > trk pos 039 on d1
13:07:32.707 > trk pos 040 on d1
13:07:32.710 > trk pos 041 on d1
13:07:32.713 > trk pos 042 on d1
13:07:32.716 > trk pos 043 on d1
13:07:32.719 > trk pos 044 on d1
13:07:32.722 > trk pos 045 on d1
13:07:32.725 > trk pos 046 on d1
13:07:32.728 > trk pos 047 on d1
13:07:32.730 > trk pos 048 on d1
13:07:32.733 > trk pos 049 on d1
13:07:32.736 > trk pos 050 on d1
13:07:32.739 > trk pos 051 on d1
13:07:32.742 > trk pos 052 on d1
13:07:32.745 > trk pos 053 on d1
13:07:32.747 > trk pos 054 on d1
13:07:32.751 > trk pos 055 on d1
13:07:32.753 > trk pos 056 on d1
13:07:32.756 > trk pos 057 on d1
13:07:32.759 > trk pos 058 on d1
13:07:32.762 > trk pos 059 on d1
13:07:32.765 > trk pos 060 on d1
13:07:32.769 > trk pos 061 on d1
13:07:32.772 > trk pos 062 on d1
13:07:32.776 > trk pos 063 on d1
13:07:32.780 > trk pos 064 on d1
13:07:32.783 > trk pos 065 on d1
13:07:32.788 > trk pos 066 on d1
13:07:32.793 > trk pos 068 on d1
13:07:33.824 > trk pos 070 on d1
13:07:33.836 > trk pos 071 on d1
13:07:33.840 > trk pos 072 on d1
13:07:33.845 > trk pos 073 on d1
13:07:33.849 > trk pos 074 on d1
13:07:33.854 > trk pos 076 on d1
13:07:34.156 > trk pos 068 on d2 <-- HERE
13:07:37.900 > disk ii enable states: 00
13:07:37.900 > stop diskII
13:07:37.900 > disk ii enable states: 00
```

Am wondering if there's some crossed wires as far as disk ][ state?

FozzTexx, 2025-03-23:
> Am wondering if there's some crossed wires as far as disk ][ state?

I'll bet it's because even though I fixed it to copy the other disk's track in without waiting for the motor to stop I didn't do anything about changing the write protect state.

So many edge cases.

closed 2025-03-24 by tschak909