FujiNetWIFI/fujinet-firmware #891: TNFS leaks memory when writing

FozzTexx, 2025-03-26:
When writing to a TNFS server the FujiNet will leak memory:

Steps to reproduce:

1. Mount DOS 3.3 System Master read-only in Disk II drive 1
2. Create new blank DOS 3.3 on TNFS server and mount as write in Disk II drive 2
3. Boot DOS 3.3
4. BRUN FID
5. Push A button so FujiNet prints current Heap on console
6. Choose COPY
  * Source S6D1
  * Dest S6D2
  * Filename: `=`
  * Prompting: `N`
7. After copy finishes push A button to get current Heap

Memory leak doesn't happen when writing to SD.

Other things that *should* also exhibit TNFS memory leak:

* Writing to a ProDOS image
* Writing via N: drive

dillera, 2026-10-03:
This issue has had no activity for about 19 months, and there have been 680 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-12:
@FozzTexx #1610 fixed a leak in the Disk II write ISR: when its 10-deep write queue was full, the dropped item's buffer was never freed. TNFS writes are slow enough to back that queue up and SD writes aren't, which matches what you saw. Can you rerun the FID copy on current firmware? If the heap holds flat I'll close this. If a debug build logs "Disk II write items dropped", writes are still being lost, just not leaked, and I'll open a separate issue for that.