FujiNetWIFI/fujinet-firmware #892: ProDOS cannot be booted from Disk II

FozzTexx, 2025-03-27:
When trying to boot ProDOS from Disk II it will click, display the splash/copyright, and then get stuck in a loop trying to move off track 1 but never quite make it.

```
09:56:52.022 > trk pos 01.0/Q004 on d1
09:56:52.060 > trk pos 01.2/Q006 on d1
09:56:52.072 > trk pos 01.1/Q005 on d1
09:56:52.072 > trk pos 01.0/Q004 on d1
09:56:52.111 > trk pos 01.2/Q006 on d1
09:56:52.122 > trk pos 01.0/Q004 on d1
09:56:52.161 > trk pos 01.2/Q006 on d1
09:56:52.173 > trk pos 01.1/Q005 on d1
09:56:52.173 > trk pos 01.0/Q004 on d1
09:56:52.212 > trk pos 01.2/Q006 on d1
09:56:52.224 > trk pos 01.0/Q004 on d1
09:56:52.262 > trk pos 01.2/Q006 on d1
09:56:52.274 > trk pos 01.1/Q005 on d1
09:56:52.274 > trk pos 01.0/Q004 on d1
09:56:52.313 > trk pos 01.2/Q006 on d1
09:56:52.324 > trk pos 01.0/Q004 on d1
09:56:52.363 > trk pos 01.2/Q006 on d1
09:56:52.375 > trk pos 01.1/Q005 on d1
09:56:52.375 > trk pos 01.0/Q004 on d1
09:56:52.414 > trk pos 01.2/Q006 on d1
09:56:52.426 > trk pos 01.1/Q005 on d1
09:56:52.426 > trk pos 01.0/Q004 on d1
09:56:52.464 > trk pos 01.2/Q006 on d1
09:56:52.476 > trk pos 01.1/Q005 on d1
09:56:52.476 > trk pos 01.0/Q004 on d1
09:56:52.515 > trk pos 01.2/Q006 on d1
09:56:52.526 > trk pos 01.0/Q004 on d1
09:56:52.565 > trk pos 01.2/Q006 on d1
09:56:52.577 > trk pos 01.1/Q005 on d1
09:56:52.577 > trk pos 01.0/Q004 on d1
09:56:52.616 > trk pos 01.2/Q006 on d1
09:56:52.628 > trk pos 01.0/Q004 on d1
09:56:52.666 > trk pos 01.2/Q006 on d1
09:56:52.678 > trk pos 01.1/Q005 on d1
09:56:52.678 > trk pos 01.0/Q004 on d1
09:56:52.717 > trk pos 01.2/Q006 on d1
09:56:52.728 > trk pos 01.0/Q004 on d1
09:56:52.767 > trk pos 01.2/Q006 on d1
09:56:52.779 > trk pos 01.0/Q004 on d1
09:56:52.818 > trk pos 01.2/Q006 on d1
09:56:52.830 > trk pos 01.0/Q004 on d1
09:56:52.869 > trk pos 01.2/Q006 on d1
```

tschak909, 2025-03-28:
Yup, I can reproduce this:

* Apple II+ 
* SoftSP v6 DIY in Grappler+
* ProDOS 2.4.3 in Disk ][ drive 1.

Exactly the same output as above.

tschak909, 2025-03-28:
Also reproduced on Apple //c with Rom0
-Thom

tschak909, 2025-03-28:
And if I let it sit on the //c long enough, I get a relocation error, with the following backtrace:
```
09:53:18.700 > trk pos 006 on d1Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 
09:53:18.701 > 
09:53:18.701 > Core  1 register dump:
09:53:18.703 > PC      : 0x4008f892  PS      : 0x00060035  A0      : 0x80082598  A1      : 0x3ffc2e20  
09:53:18.704 > A2      : 0x3ffc2e3c  A3      : 0x3f400928  A4      : 0x00000020  A5      : 0x3ffc2e3c  
09:53:18.706 > A6      : 0xbad00bad  A7      : 0xbad00bad  A8      : 0x00000000  A9      : 0x3ffbaa14  
09:53:18.708 > A10     : 0x3ffc9dc0  A11     : 0x00060023  A12     : 0x00000001  A13     : 0x003fffff  
09:53:18.710 > A14     : 0x00000003  A15     : 0x00060623  SAR     : 0x00000003  EXCCAUSE: 0x00000007  
09:53:18.712 > EXCVADDR: 0x00000000  LBEG    : 0x4008f88c  LEND    : 0x4008f8a8  LCOUNT  : 0x00000001  
09:53:18.714 > 
09:53:18.714 > 
09:53:18.714 > Backtrace: 0x4008f88f:0x3ffc2e20 0x40082595:0x3ffc2e30 0x4008268f:0x3ffc2e80 0x4008303e:0x3ffc2eb0 0x4008628d:0x3ffc2ef0 0x40087426:0x3ffafd50 0x40084b33:0x3ffafd70 0x400985c1:0x3ffafd90
09:53:18.883 > 
09:53:18.883 >   #0  0x4008f88f:0x3ffc2e20 in memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:176
09:53:18.883 >   #1  0x40082595:0x3ffc2e30 in iwm_diskii_ll::fakebit() at lib/bus/iwm/iwm_ll.cpp:1127
09:53:18.883 >   #2  0x4008268f:0x3ffc2e80 in encode_rmt_bitstream(void const*, rmt_item32_t*, unsigned int, unsigned int, unsigned int*, unsigned int*, int) at lib/bus/iwm/iwm_ll.cpp:1050 (discriminator 2)
09:53:18.883 >   #3  0x4008303e:0x3ffc2eb0 in fn_rmt_driver_isr_default(void*) at lib/hardware/fnRMTstream.cpp:735
09:53:18.883 >   #4  0x4008628d:0x3ffc2ef0 in _xt_lowint1 at /home/thomc/.platformio/packages/framework-espidf@3.50002.230601/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118
09:53:18.883 >   #5  0x40087426:0x3ffafd50 in spi_flash_op_block_func at /home/thomc/.platformio/packages/framework-espidf@3.50002.230601/components/spi_flash/cache_utils.c:133
09:53:18.883 >   #6  0x40084b33:0x3ffafd70 in ipc_task at /home/thomc/.platformio/packages/framework-espidf@3.50002.230601/components/esp_system/esp_ipc.c:87
09:53:18.883 >   #7  0x400985c1:0x3ffafd90 in vPortTaskWrapper at /home/thomc/.platformio/packages/framework-espidf@3.50002.230601/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

FozzTexx, 2025-03-28:
> 09:53:18.883 >   #0  0x4008f88f:0x3ffc2e20 in memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:176
> 09:53:18.883 >   #1  0x40082595:0x3ffc2e30 in iwm_diskii_ll::fakebit() at lib/bus/iwm/iwm_ll.cpp:1127

There is no `memcpy()` call in `fakebit()` so that seems a bit odd.

FozzTexx, 2025-03-28:
What seems to be happening is that the FujiNet is missing steps. ProDOS steps the head down just fine by rolling the stepper waveform, with around 4 milliseconds per step. Suddenly the FujiNet has a delay of 40 milliseconds every 4th step and so it thinks the stepper is going the other way and does D=2.

```
11:44:22.195 > trk pos 03.0/Q012=Q003 idx=7 D=-1 P=0001/1 on d1
11:44:22.199 > trk pos 02.3/Q011=Q003 idx=7 D=-1 P=1001/9 on d1
11:44:22.202 > trk pos 02.2/Q010=Q255 idx=7 D=-1 P=1000/8 on d1
11:44:22.204 > trk pos 02.1/Q009=Q002 idx=7 D=-1 P=1100/12 on d1
11:44:22.204 > Copy track: 00.2/Q009
11:44:22.210 > trk pos 02.0/Q008=Q002 idx=7 D=-1 P=0100/4 on d1
11:44:22.214 > trk pos 01.3/Q007=Q002 idx=7 D=-1 P=0110/6 on d1
11:44:22.218 > trk pos 01.2/Q006=Q255 idx=7 D=-1 P=0010/2 on d1
11:44:22.222 > trk pos 01.1/Q005=Q001 idx=7 D=-1 P=0011/3 on d1
11:44:22.222 > Copy track: 00.1/Q005
11:44:22.225 > trk pos 01.0/Q004=Q001 idx=7 D=-1 P=0001/1 on d1
11:44:22.261 > trk pos 01.2/Q006=Q255 idx=2 D=2 P=0010/2 on d1
11:44:22.271 > trk pos 01.1/Q005=Q001 idx=7 D=-1 P=0011/3 on d1
11:44:22.271 > Copy track: 00.1/Q005
11:44:22.274 > trk pos 01.0/Q004=Q001 idx=7 D=-1 P=0001/1 on d1
11:44:22.311 > trk pos 01.2/Q006=Q255 idx=2 D=2 P=0010/2 on d1
11:44:22.321 > trk pos 01.1/Q005=Q001 idx=7 D=-1 P=0011/3 on d1
11:44:22.321 > Copy track: 00.1/Q005
11:44:22.324 > trk pos 01.0/Q004=Q001 idx=7 D=-1 P=0001/1 on d1
11:44:22.362 > trk pos 01.2/Q006=Q255 idx=2 D=2 P=0010/2 on d1
11:44:22.372 > trk pos 01.1/Q005=Q001 idx=7 D=-1 P=0011/3 on d1
11:44:22.372 > Copy track: 00.1/Q005
11:44:22.375 > trk pos 01.0/Q004=Q001 idx=7 D=-1 P=0001/1 on d1
11:44:22.412 > trk pos 01.2/Q006=Q255 idx=2 D=2 P=0010/2 on d1
```

tschak909, 2025-03-28:
wow.

FozzTexx, 2025-03-29:
Hooked up logic analyzer to see what ProDOS is actually sending when it goes into the loop. It really does have a 40ms delay and has stopped stepping.

[image]

FozzTexx, 2025-03-29:
Just to confirm logic analyzer is hooked up correctly, here's a capture during the initial boot just before the splash screen. 

[image]

robjustice, 2025-03-29:
I had a try with this disk image. the first time it booted up ok. Then I did a few more and its gives me inconsistent behavior. works sometimes, and other times gives the looping symptoms. On a IIe enhanced with a db19 controller and the latest rc1.5 build

dillera, 2026-10-03:
This issue has had no activity for about 18 months, and there have been 680 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

FozzTexx, 2026-10-03:
This bug most definitely still exists.

dillera, 2026-10-03:
Thanks for confirming this still happens on current firmware — keeping it open.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-12:
Still valid per @FozzTexx. One concrete thing from @tschak909's backtrace: `fakebit()` declares `MC3470[]` as a non-static local `const`, so every call copies it out of flash rodata (that's the `memcpy`), and `fakebit()` runs in the RMT ISR. If a flash op has the cache disabled at that moment, it panics, and that happens on blank quarter-tracks like Q006 here. `static const DRAM_ATTR` fixes that crash; split out as #1894. It doesn't explain the boot loop. The logic analyzer shows ProDOS itself pausing and re-seeking, which looks like it's failing to read track 1 and recalibrating, not FujiNet missing steps.