FujiNetWIFI/fujinet-firmware #907: Apple II: Possible conflict with High Speed SCSI card

FujiMan66, 2025-04-05:
Apple IIe - Fujinet Firmware 1.5
Slot 2: Super Serial Card
Slot 5: Grappler+SPv6
Slot 6: Floppy Controller
Slot 7: High Speed SCSI card

TNFS hosts and Drive slots can be populated from browser.
FujiNet connect to wireless access point, but has the following issues.

    Screen shows list of 16 access points with no text with 3 asterisks
    FujiNet config information screen is non-populated.
    SmartPort device list reports no devices.
    SD card does not show contents.
    Removal of HS SCSI allows FN to return to proper function.

FozzTexx, 2025-04-16:
Able to reproduce without High Speed SCSI card. Instead I configured a second SoftSP card to use a Disk II controller in another slot. With the FN connected to the Disk II card which is used by the SP controller in the lower slot, CONFIG will boot but will fail to locate FujiNet.

Also see by running my SmartPort.Exorciser test that fujinet-lib fails to locate the FujiNet.

@markjfisher Does fujinet-lib stop after finding the first SmartPort controller?

FozzTexx, 2025-04-17:
> [@markjfisher](https://github.com/markjfisher) Does fujinet-lib stop after finding the first SmartPort controller?

Confirmed that fujinet-lib stops after finding the first SP card. Tried out the original C version (which was conveniently left as a comment in `sp_init.s`, thanks!) and it is able to progress and find the FujiNet. See attached screen shot.

[image]

markjfisher, 2025-04-17:
yeah, sorry, I must have broken the code when trying to optimize it in asm (again).
It's not supposed to stop after scanning the 1 slot, it's supposed to find it no matter where.

Can you try making a 1 line change and testing it:
[image]

Add the "lda _sp_network" as follows after the jsr:
```
.endif
        jsr     _sp_get_network_id

        lda     _sp_network    ; Check the actual network ID
        bne     found_network
```

I think that may be needed to fix this rather than going with the C code.

FozzTexx, 2025-04-18:
https://github.com/FujiNetWIFI/fujinet-lib/pull/25 fixes the problem with fujinet-lib stopping at the first SmartPort card.

FozzTexx, 2025-04-19:
@FujiMan66 can you test this: https://github.com/FujiNetWIFI/fujinet-firmware/pull/913

PR #913 (2025-04-20): [apple2] Fixes #907: update CONFIG to use fujinet-lib 4.7.5.
Fixes #907 issue where searching for the FujiNet was not checking all available SmartPort interfaces.
commit 8d6e6918 (2025-04-20): [apple2] Fixes #907: update CONFIG to use fujinet-lib 4.7.5. (#913)
closed 2025-04-20 by tschak909