FujiNetWIFI/fujinet-firmware #914: Eject (IO15) sensing not working

tschak909, 2025-04-22:
Reported by John Lundy via Facebook:

It has been a while since I’ve used ADAM FujiNet or had to build more units for stock. I recently built more FujiNet Mini units, and during testing, noticed the SD card eject/insert sense is no longer functioning after booting. The card will only work when booting fresh or after a reset. I can see I/O15 working fine with a low when a card is inserted and high when removed. After doing some troubleshooting, this functionality hasn’t worked since the release of v1.4. Dropping all the way back to v1.3 will work as expected. The serial debug shows absolutely nothing when the card is removed/inserted, as if I/O15 is no longer being monitored for any state change at all in code now.

mozzwald, 2025-04-22:
I'm sure this is my fault from when I changed the board detection logic. Will take a look

tschak909, 2025-04-22:
@mozzwald this should fix it.

closed 2025-04-22 by mozzwald