FujiNetWIFI/fujinet-firmware #917: Improve CP/M file performance.

tschak909, 2025-04-24:
File I/O is slow in CP/M.

This is because for each BIOS read sector:

* The file is opened
* The sector is read
* The file is closed.

A similar pattern is done for the other methods I list below, it's very atomic, but very slow.

Calls to OpenFile() in lib/runcpm/disk.h call _sys_openfile(): 
https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/lib/runcpm/abstraction_fujinet.h#L153 and 
read sequential: https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/lib/runcpm/abstraction_fujinet.h#L223
and write sequential: https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/lib/runcpm/abstraction_fujinet.h#L263
and read random: https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/lib/runcpm/abstraction_fujinet.h#L293
and write random: https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/lib/runcpm/abstraction_fujinet.h#L339

This needs to be extended to hold onto the file handle, and not open the file again if it has already been opened, thus avoiding the file_open call if needed, and finding out when to best close the file when it is no longer needed (so we don't leak file descriptors! A timer?)

-Thom

dillera, 2026-10-03:
This issue has had no activity for about 18 months, and there have been 641 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-12:
Still applies on current master. All three abstractions (`abstraction_fujinet.h`, `abstraction_fujinet_apple2.h`, `abstraction_network_protocol.h`) still open, seek and close the file for every 128-byte record. Any handle caching should go in once, after those are merged into one, rather than three times. Keeping this open.