FujiNetWIFI/fujinet-firmware #918: Improve CP/M printer performance.

tschak909, 2025-04-24:
CP/M printer performance is slow.

This is because we are sending each individual character to the virtual printer, one character at a time conforming to the CP/M BIOS function.

see: https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/lib/runcpm/cpm.h#L637
and: https://github.com/FujiNetWIFI/fujinet-firmware/blob/master/lib/device/sio/printer.cpp#L115

The buffer in question happens to be 320 bytes long, so up to 320 bytes can be sent to the printer in print_from_cpm(), which would vastly speed up the process, and using a timer to indicate when to flush the buffer (assuming less than 320 bytes is in it)

-Thom

dillera, 2026-10-03:
This issue has had no activity for about 18 months, and there have been 641 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-12:
Still applies on current master for SIO, RS232, DriveWire and CX16, which still call `process()` per character. Apple and Mac already buffer to CR/LF/80 chars (a3266766b) but have no timer flush, so a trailing partial line waits for the next newline. IEC's `print_from_cpm` is still a stub. Since it's now one function copied per bus, the buffered version should be hoisted once rather than copied again. Keeping this open.