FujiNetWIFI/fujinet-firmware #919: Can't Mount disks as R/W in Apple II Config UI

pmontelo, 2025-04-27:
FujiNet version	v1.5.0 2025-04-02 10:07:01 (APPLE)
If you use the Config UI on the Apple II to mount disks as R/W they will be read-only.
A work-around is to mount them in the Web UI or select "MOUNT ALL DISKS" in the Web UI.

[image]

[image]

FozzTexx, 2025-08-12:
Is this still an issue with the latest nightly? https://github.com/FujiNetWIFI/fujinet-firmware/releases/tag/nightly_fn_pc_v1.5.0%2Bgit-58ec8f2c

pmontelo, 2025-08-12:
I am away from home until September, I will test when I get back. Thank you for opening a bug

closed 2025-08-24 by FozzTexx