FujiNetWIFI/fujinet-firmware #945: Time zones stopped working after FW update

cosi1, 2025-06-08:
Hey, FujiNet team!

Right after updating the firmware to version 1.5.0 I noticed that `apetime` yields UTC time regardless of the timezone setting in FujiNet's control panel (mine says "CET-1CEST,M3.5.0,M10.5.0/3", but I've tested several different TZs with the same result).

I have `apetime` in my autoexec.bat, so I'm sure the issue occurred during the update, but I can't tell if it's the updated firmware that caused the problem.

cosi1, 2025-06-10:
It says:
```[19:12:31] CF: 45 93 ee a0 68
[19:12:31] ACK!
[19:12:31] ->SIO write 6 bytes
[19:12:31] COMPLETE!```
Curiously (or maybe not), when I run `getclk`, I'm getting a completely different output...

markjfisher, 2025-06-10:
The last 2 bytes are minutes and seconds, hence why they are showing different between runs.

However, I see that lines 3 and 5 are the same in both runs (spartados and bare).
That tells me your fujinet would normally send your timezone's time correctly.
It even shows your FN's current timezone in the 2nd paragraph and 4th paragraph, and they agree with the 1st paragraph which I manually set in the program to check.

I'm afraid I don't see anything wrong with this. Your fujinet is sending the correct APETIME bytes for your timezone, as shown on line 3 of both runs being identical to line 5 of both runs.

cosi1, 2025-06-17:
Glad I could help! 😄👍