FujiNetWIFI/fujinet-firmware #971: On Apple2GS config doesn't work booting on 1.5, but does work from 1.4

cathrynm, 2025-07-05:
Curious if anyone else sees this.  On IIGS,  I have 1.5 and with this, config works if I boot to GSOS with CFFA 3000, and then use the control panel version of Fujinet Config.  But, if I try booting from Fujinet, it shows no wifi networks, and the Config option shows nothing. It looks like the app is loading but it's not talking to hardware.

I downgraded back to 1.4, and now it shows config find on boot.  (I know this is a weird one.)

I am able to generally use 1.5 if I manage it from the website.

I also have AppleSqueezer, and turned all of its options off so it's running as normal Fast mode on IIGS  for this just in case (though I'd rather not pry it out as I don't want to stress my CPU socket).  Don't think this is related, but here for completeness.

FozzTexx, 2025-07-05:
Does this happen with the latest nightly?

nriley, 2025-07-05:
With a similar setup nightly won't boot CONFIG at all. I haven't tested recently but this setup has been broken as far back as at least late 2024. I did bisect this at some point to find the commit that broke things but I recall I may have only posted my results on Discord and I'm not sure if they're findable any more…

cathrynm, 2025-07-05:
It looks related to having CFFA-3000 enabled in slot 7.  Issue shows up with 'Your Card' enable in slot 7.

cathrynm, 2025-07-05:
Oh, and it is working on v1.5.0.10f3f7.   Going to poke around this for a little bit, see how it goes.

nriley, 2025-07-05:
@cathrynm I also have the CFFA3000 in slot 7 on my IIgs. If I remove all the SmartPort disk assignments from the CFFA3000 configuration then CONFIG boots OK; it's only if there are SmartPort devices in slot 7 that it breaks. This used to work if you set the boot slot to 5.

cathrynm, 2025-07-05:
When I run modem-and-cpm.po on the daily, modem works, but cpm doesn't. This is unrelated to whether or not CFA-3000 is enabled.  CPM works fine for me on the 1.5. (Maybe also unrelated dev random.)

dillera, 2026-10-03:
More information was asked for here about 14 months ago and there has been no reply, so closing this for now. If you can still reproduce this on a current firmware build, please reopen it (or open a new issue) with the firmware version, the platform, and the steps.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

closed 2026-10-03 by dillera