FujiNetWIFI/fujinet-firmware #987: GSOS Fails to load with currently master

ppuskari, 2025-08-01:
Been trying to get GSOS to work fully again like it was able to in 1.4  Still no luck with 1.5 and current master.

Grab a copy of https://www.whatisthe2gs.apple2.org.za/files/harddrive_image.zip

put the expanded hdv volume on tnfs and or sd card.  Both crash during boot.

All the way back to 1.5 it will crash most of the time during the boot process.  If it happens to get to the desktop then the boot volume is LOCKED.  Regardless of the Write option being set.  Other volumes mounted will be listed as write enabled properly.

1.4 boot fine the write locking issue shows immediately upon the 1.4+ commit  fb2aa19

1.4+ commit febf356 works for gsos launching and correct r/w enablement.
1.4+ commit fb2aa19 where Disk II write support is added is where it breaks.  

I have not found any commit post febf356 where gsos still works correctly.  It's either loading in READONLY mode regardless of mount setting or crashing during boot.

ppuskari, 2025-08-01:
disabling printf debug statements didn't help this time either.  I'm going to rebuild with more verbose logging and see what I can find.

FozzTexx, 2025-08-01:
I can't seem to boot that disk at all. Gets the "Welcome to the IIgs" thermometer, reads from the disk for a while, but then the last activity from the FujiNet is
```
14:18:43.371 > handling read block command
14:18:43.372 > Drive 81  Read block 000681
14:18:43.372 > iwm_readblock NORMAL READ
14:18:43.374 > 
14:18:43.374 > sending block packet ...
14:18:43.405 > handling read block command
14:18:43.405 > Drive 81  Read block 000682
14:18:43.405 > iwm_readblock NORMAL READ
14:18:43.407 > 
14:18:43.407 > sending block packet ...
14:18:43.766 > disk ii [1] enable states: 01
14:18:43.766 > stop diskII
14:18:43.766 > handling status command
14:18:43.767 > Sending DIB Status for device 0x81
14:18:43.767 > DIB number of blocks 65535
14:18:43.770 > 
14:18:43.770 > FUJINET_DISK_0: Sending DIB reply with status: 0xF8
```

ppuskari, 2025-08-01:
That's basically what I'm seeing too.  I do know that this image does have the BRAM loader enabled in it.  Where if it checks the IIgs active bram state compared to what is stored state of the NDA.  It will set the bram up and then reboot.  I wonder if that is havng some kind of issue..  I'll disable the NDA and try again.  But I have to do that from 1.4 where the image is write enabled... smh

ppuskari, 2025-08-02:
Revert Commit 76b76fa7b50d395f55944e7b9090911618fddc01      I modified my local source to do the mount then the read/write set and now GSOS is happy....  Need to check 5.25" disk images next.  Thomas made the change and it looks like it should work but I guess it doesn't for GSOS and the first/boot volume mounted.

FozzTexx, 2025-08-02:
> Revert Commit [76b76fa](https://github.com/FujiNetWIFI/fujinet-firmware/commit/76b76fa7b50d395f55944e7b9090911618fddc01)

No change, still gets the thermometer half way and then stops reading the disk. Thermometer keeps filling because of IRQs but there's no more disk activity.

```
06:45:54.147 > handling status command
06:45:54.147 > Sending DIB Status for device 0x81
06:45:54.147 > DIB number of blocks 65535
06:45:54.147 > 
06:45:54.147 > FUJINET_DISK_0: Sending DIB reply with status: 0xF8
06:46:22.806 > 
```

ppuskari, 2025-08-02:
Maybe I can send my build zip file and the image I’m using.  It does appear that the particular 6.0.1 gsos image I’m using has most of the drivers turned off I noticed late last night.  So maybe I’m not running into something now on his base image I turned off.

I’ll test more this afternoon too.   I am compiling off a git pull origin master then changing the Fuji.cpp file and put the mount before the read only if statement and commented out the later one.

FozzTexx, 2025-08-02:
Verified that the image is good and will boot from SCSI on my IIgs, so it's not my IIgs that is the problem.

FozzTexx, 2025-08-02:
Going back to v1.4.0 and same thing, doesn't boot
```
09:43:49.062 > handling status command
09:43:49.062 > Target Device: 81
09:43:49.062 > Sending **DIB** Status
09:43:49.063 > DIB number of blocks 65535
09:43:49.063 > 
09:43:49.063 > FUJINET_DISK_0: Sending DIB reply with status: F8
```

FozzTexx, 2025-08-02:
Went all the way back to v1.3.0 and exact same problem:
```
09:47:32.456 > handling status command
09:47:32.457 > Target Device: 81
09:47:32.457 > ******** Sending DIB! ********
09:47:32.457 > send_status_dib_reply_packet status = F8
09:47:32.459 > 
09:47:32.459 > DIB number of blocks 65535
09:47:32.459 > f8 ff ff 00
```
The FujiNet doesn't crash, the IIgs just stops asking to read sectors. I don't know if it's because the FujiNet is interpreting a command wrong and sending back the DIB, but that's the impression I'm getting.

ppuskari, 2025-08-02:
How much ram does your rom 01 have?  This image needs about 2 meg to even come up.   I just relaunched my build from scratch and things still work off current master and the one line moved befor the if statement in fuji.cpp

ppuskari, 2025-08-02:
Want to try my fujinet firmware image?  Here's what I'm currently running from master today plus the one line move in fuji.cpp I'll attach it here if it will allow .zip  or I'll put it on discord.

FozzTexx, 2025-08-02:
> I'll attach it here if it will allow .zip

That should work fine.

Here is the image I found on Discord.

[fujinet-APPLE-v1.5.0.zip](https://github.com/user-attachments/files/21561041/fujinet-APPLE-v1.5.0.zip)

This one works and my IIgs boots all the way into the desktop!

Can you make a branch with your changes and push it up to github somewhere? Not necessary to open a PR yet, but if you want to open a draft PR with your changes, that's fine too.

closed 2025-08-14 by FozzTexx