FujiNetWIFI/fujinet-firmware #996: Fujinet freezing when using Copy II Plus

matthew-apa, 2025-08-17:
Hello, I'm using the FujiNet with Wifi with my Apple IIe, and I've flashed the firmware to 1.5. I'm running a TNFS server (downloaded from this wiki) on my Mac desktop with OS Ventura 13.7.3. My SmartPort card (Soft SP Card) is in Slot 7 of my Apple IIe, and the FujiNet unit itself is connected to an Apple I/O Controller Card which is in Slot 6 of my Apple IIe. My two real floppy drives are connected to a Disk II Controller Card which is in Slot 5 of my Apple IIe.  I have ProDOS 8 v2.4.3 mounted on Fujinet Slot 4 (SmartPort). Through ProDOS I am using Copy II Plus to make real floppy disks from .dsk or .po images stored on the SD card of the FujiNet.

About half the time, I can successfully use Copy II Plus to make floppy disks. But the other half of the time, Copy II Plus just freezes during the "read" phase and eventually returns a read error on all tracks. The only solution to this seems to be to reset the FujiNet, reboot, and re-mount everything. Then sometimes it works, sometimes it still doesn't.

What am I missing, or what am I doing wrong?

Thank you,
Matthew

tschak909, 2025-08-17:
Can you provide debug output from the monitor in the FujiNet flasher while
you perform the copy?

matthew-apa, 2025-08-17:
I just edited my post to correct that the floppy .dsk and .po images are on my Mac desktop and I am using my TNFS server (the floppy images are not on my SD card as I initially posted).

Yes, thank you, I will do as you suggest and post it here.

matthew-apa, 2025-08-17:
[error1fujinet.pdf](https://github.com/user-attachments/files/21824878/error1fujinet.pdf)
[error1tnfs.pdf](https://github.com/user-attachments/files/21824877/error1tnfs.pdf)
[error2fujinet.pdf](https://github.com/user-attachments/files/21824879/error2fujinet.pdf)
[error2tnfs.pdf](https://github.com/user-attachments/files/21824876/error2tnfs.pdf)

matthew-apa, 2025-08-17:
Here are the FujiNet and TNFS server monitor outputs - for two difference instances of the problem. The first time ("error 1") Copy II Plus showed a read error for all tracks, and the second time ("error 2") Copy II Plus just froze up and the FujiNet went on some sort of infinite loop until I reset the Apple IIe. Thank you!!

tschak909, 2025-08-17:
Thanks, I've let the people in #apple-ii channel on our discord know. 
-Thom

matthew-apa, 2025-08-21:
Hello, any other thoughts on this, such as a disk copy program I might be able to use that will work with FujiNet? Thank you!

FozzTexx, 2025-08-24:
It's this:

https://github.com/FujiNetWIFI/fujinet-firmware/issues/892

dillera, 2026-10-03:
This issue has had no activity for about 13 months, and there have been 581 commits to the firmware since it was opened. Is this still happening on a current firmware build? If so, please add the firmware version and platform; if there is no update it may be closed in a later cleanup.

---
_Drafted by the FujiNet triage harness (Jev jev-1.13.0); posted by dillera after human review._

RichStephens, 2026-10-12:
@matthew-apa Both of your logs show the image being read through FujiNet's Disk II emulation, and error2 has the same track 1 loop as #892, so this is that bug; it isn't the copy program or your setup. Closing as a duplicate so it's tracked in one place; follow #892 for the fix.

closed 2026-10-12 by RichStephens